yanshay/SpoolEase #5: Bambu FIlaments

matthewtrains, 2025-02-25:
It seems Bambu filaments are not showing up on the display, is this expected?

yanshay, 2025-02-25:
> It seems Bambu filaments are not showing up on the display, is this expected?

I don't use Bambu filaments so didn't check. I should have one that came with the printer so will check tomorrow. No reason it won't work but maybe needs some special handling in their case.

@Traisuma was any of your spools from Bambu?

yanshay, 2025-02-25:
> I ended up screwing up the NFC reader, so i ordered two more to reattempt. 

Why do you think you screwed up the NFC reader?

Traisuma, 2025-02-25:
Yeah, 2 are officals in my screenshot

matthewtrains, 2025-02-26:
I think you may need to add some code that checks for loaded filaments once the unit powers on, but ill let the picture do the explaining for the screw up lol (Still new to soldering, i know how to avoid my mistake next time.)

[image]

matthewtrains, 2025-02-26:
My screw up started once i accidently broke off what i belive to be a capicator. Next time im just going to use the included pins 

[image]

matthewtrains, 2025-02-26:
It started when i accidently got some solder on what i belive to be a cap and i made the mistake of trying to pick it off which took the cap (i thats what that is) off the board

Traisuma, 2025-02-26:
Hey, i just double checked in the morning.
I have 2 Sunlu and 2 Bambu inside.
My SpoolEase recognizes all 4 Spools with the correct K value & Color.

Even after Restarting the SpoolEase and restarting the Printer.

yanshay, 2025-02-26:
@matthewtrains What printer model do you have?

matthewtrains, 2025-02-26:
X1C

matthewtrains, 2025-02-26:
If i go to manually edit the external spool, it does update but thats about it, i will get the new NFC boards tommorow

yanshay, 2025-02-26:
Can you please explain in more details what you see. 

Specifically, when you restart SpoolEase and don't see on SpoolEase some filaments, and you start a **new instance** of a slicer (so close all slicers open and open Bambustudio), do you see the filament in the slicer?

yanshay, 2025-02-26:
If you can send photos of the display that's great.
Also of what's showing on the main filament view of both EasySpool and the Slicer.

I'm working on something that will be able to let you have access to more detailed logs and update the software so I can troubleshoot the issue (if you'll be able to spend the time when at home).
It's an issue that should be resolvable. Probably due to X1 behaving a bit different.

BTW - How many AMS units do you have? More than one? That could also be an expected issue which I'm fixing.

matthewtrains, 2025-02-26:
So the slicer displays all the filaments, and SpoolEase displays them all as empty.

[image]

[image]

Update: apparently it decided to pull the external spool color and my 2nd spool but not the 3rd or 4th

matthewtrains, 2025-02-26:
This is whats set in the slicer vs what the device saw
[image]

its not a perfect picture as i had my brother point a camera at it but here is what SpoolEase sees 

[image]

matthewtrains, 2025-02-26:
But im at a point where i cant do anything until im home, since i tried to have him enable the web config and i think he accidently clicked the reset wifi as he was scrolling down, its kinda easy to accidently press

yanshay, 2025-02-26:
No worries, there's another issue with X1C very similar. I have logs and will troubleshoot and you'll hopefully enjoy the solution once that's hopefully succeed.

matthewtrains, 2025-02-26:
Sounds good, cant wait

yanshay, 2025-02-26:
Let me know when you can try some patch, it is not final but want to see how it functions on your printer.

David8288, 2025-02-28:
Hi, 
I also have an X1C and am still waiting for the parts from China. 
As soon as everything arrives, I would also like to offer me to test your great project.
BR David

yanshay, 2025-02-28:
> Hi, I also have an X1C and am still waiting for the parts from China. As soon as everything arrives, I would also like to offer me to test your great project. BR David

Happy to hear that!

yanshay, 2025-03-01:
Issues resolved, closing

closed 2025-03-01 by yanshay