yanshay/SpoolEase #13: Encoding Tag Failed

MagicSven81, 2025-03-06:
First i want to say thank you for this great project.

I built the SpoolEase and double checked all steps.
I have a connection to my wlan and printer and i can see the spools insert in my AMS
If i change a slot in Bambu Lab, one second later i can see it in SpoolEase.

But if i try to programm the NFC Tag (ntag215)
i will get the message: **Encoding Tag Failed**

i also tried the original NFC Card and NFC Chip which will be delivered with the NFC Tag Reader,
but the same message appears.

Where is my mistake?

gn0m3r4l, 2025-03-06:
I got the same problem. I soldered wires to pins correctly but not the pins
to the NFC board.
For me, adding some soldering tin on the backside of the pins, solved the
problem

MagicSven81, 2025-03-06:
Problem solved - the 8pin connector has switched yellow and green wire.
After correcting this, it works - thx

closed 2025-03-06 by MagicSven81