yanshay/SpoolEase #14: Device freezes after it connects to printer

alien2108, 2025-03-07:
**Describe the bug**
After device is configured and all is OK device connects to the printer and then freezes. Nothing works - not even scrolling between displays.

**To Reproduce**
Steps to reproduce the behavior:
Always the same for me

LOGS:

[23:33:08]INFO - [wifi.rs:390] Still waiting for an IP address
[23:33:08]INFO - [wifi.rs:352] Received IP: 192.168.3.128/24
[23:33:08]INFO - [framework.rs:279] Checking Firmware Version Over the Air
[23:33:08]INFO - [ota.rs:53] Resolving Dns
[23:33:08]INFO - [bambu.rs:1363] From Bambu MQTT - got IP
[23:33:08]INFO - [bambu.rs:1370] No Printer IP configured, discovering Printer
[23:33:08]DEBUG - [bambu.rs:1399] Waiting for SSDP UDP
[23:33:08]INFO - [bambu.rs:1439] Discovered Printer at 192.168.3.20
[23:33:08]INFO - [bambu.rs:1440] Printer named 'Bambu X1C'
[23:33:08]INFO - [my_mqtt.rs:331] Connecting to Printer 192.168.3.20:8883
[23:33:10]INFO - [ota.rs:80] Resolved DNS for raw.githubusercontent.com [Ipv4(185.199.108.133)]
[23:33:10]INFO - [ota.rs:53] Fetching firmware metadata
[23:33:11]INFO - [ota.rs:140] Firmware metadata:
[23:33:11]filename = "SpoolEase-0.1.7.bin"
[23:33:11]version = "0.1.7"
[23:33:11]filesize = 3254816
[23:33:11]crc32 = "e9d79372"
[23:33:11]
[23:33:11]INFO - [ota.rs:65] Current firmware version 0.1.7 is up to date
[23:33:36]INFO - [my_mqtt.rs:347] Connected to Printer
[23:33:37]INFO - [my_mqtt.rs:371] Establishing TLS connection with Printer
[23:33:37]INFO - [my_mqtt.rs:380] TLS connection with Printer established
[23:33:37]INFO - [my_mqtt.rs:382] Establishing MQTT connection with Printer
[23:33:37]INFO - [my_mqtt.rs:391] MQTT connection with Printer established
[23:33:37]INFO - [my_mqtt.rs:393] Subscribing to Printer reports
[23:33:37]WARN - [my_mqtt.rs:125] Suback received with Pid(1), []
[23:33:37]INFO - [my_mqtt.rs:401] Subscription to Printer reports confirmed
[23:33:37]INFO - [view_model.rs:482] ------------------------------------------------------------------
[23:33:37]INFO - [view_model.rs:483] Startup completed successfully
[23:33:37]INFO - [view_model.rs:484] ------------------------------------------------------------------
[23:33:37]DEBUG - [bambu.rs:741] MQTT Publish: {
[23:33:37]  "pushing": {
[23:33:37]    "command": "pushall"
[23:33:37]  }
[23:33:37]}
[23:33:37]DEBUG - [bambu.rs:1291] MQTT Receive: Print { print: PrintData { ams: Some(PrintAms { ams: Some([PrintAmsData { id: "1", humidity: "2", tray: [PrintTray { id: 0, k: None, cali_idx: Some(-1), tray_info_idx: Some("GFG99"), tray_type: Some("PETG"), tray_color: Some("7C4B00FF"), nozzle_temp_max: Some(270), nozzle_temp_min: Some(220) }, PrintTray { id: 1, k: None, cali_idx: Some(-1), tray_info_idx: Some("GFL95"), tray_type: Some("PLA"), tray_color: Some("2850E0FF"), nozzle_temp_max: Some(240), nozzle_temp_min: Some(190) }, PrintTray { id: 2, k: None, cali_idx: Some(-1), tray_info_idx: Some("GFG99"), tray_type: Some("PETG"), tray_color: Some("161616FF"), nozzle_temp_max: Some(270), nozzle_temp_min: Some(220) }, PrintTray { id: 3, k: None, cali_idx: Some(-1), tray_info_idx: Some("GFG99"), tray_type: Some("PETG"), tray_color: Some("FFF144FF"), nozzle_temp_max: Some(270), nozzle_temp_min: Some(220) }] }]), ams_exist_bits: Some("2"), tray_exist_bits: Some("f0"), tray_is_bbl_bits: Some("f0"), tray_read_done_bits: Some("f0"), tray_reading_bits: Some("0") }), vt_tray: Some(PrintTray { id: 254, k: None, cali_idx: Some(-1), tray_info_idx: Some("GFB99"), tray_type: Some("ABS"), tray_color: Some("FFFFFFFF"), nozzle_temp_max: Some(280), nozzle_temp_min: Some(240) }), command: Some("push_status"), sequence_id: Some("2021"), nozzle_temp_max: None, nozzle_temp_min: None, tray_color: None, tray_id: None, ams_id: None, cali_idx: None, tray_info_idx: None, tray_type: None, reason: None, result: None, nozzle_diameter: Some("0.4"), filament_id: None, filaments: None } }
[23:33:37]DEBUG - [bambu.rs:645] -> Message : sequence_id = "2021"
[23:33:37]DEBUG - [bambu.rs:655]              push_status message
>

FREEZES AFTER THIS

alien2108, 2025-03-07:
Rebooted from console and now logs show a bit more before freeze:

[23:37:30]INFO - [ota.rs:65] Current firmware version 0.1.7 is up to date
[23:37:32]INFO - [my_mqtt.rs:347] Connected to Printer
[23:37:32]INFO - [my_mqtt.rs:371] Establishing TLS connection with Printer
[23:37:32]INFO - [my_mqtt.rs:380] TLS connection with Printer established
[23:37:32]INFO - [my_mqtt.rs:382] Establishing MQTT connection with Printer
[23:37:32]INFO - [my_mqtt.rs:391] MQTT connection with Printer established
[23:37:33]INFO - [my_mqtt.rs:393] Subscribing to Printer reports
[23:37:33]WARN - [my_mqtt.rs:125] Suback received with Pid(1), []
[23:37:33]INFO - [my_mqtt.rs:401] Subscription to Printer reports confirmed
[23:37:33]INFO - [view_model.rs:482] ------------------------------------------------------------------
[23:37:33]INFO - [view_model.rs:483] Startup completed successfully
[23:37:33]INFO - [view_model.rs:484] ------------------------------------------------------------------
[23:37:33]DEBUG - [bambu.rs:741] MQTT Publish: {
[23:37:33]  "pushing": {
[23:37:33]    "command": "pushall"
[23:37:33]  }
[23:37:33]}
[23:37:33]DEBUG - [bambu.rs:1291] MQTT Receive: Print { print: PrintData { ams: None, vt_tray: None, command: Some("extrusion_cali_get"), sequence_id: Some("1"), nozzle_temp_max: None, nozzle_temp_min: None, tray_color: None, tray_id: None, ams_id: None, cali_idx: None, tray_info_idx: None, tray_type: None, reason: Some("fila_device_extrusion_get_all error"), result: Some("fail"), nozzle_diameter: Some("0.8"), filament_id: Some(""), filaments: Some([]) } }
[23:37:33]DEBUG - [bambu.rs:645] -> Message : sequence_id = "1"
[23:37:33]DEBUG - [bambu.rs:694]              extrusion_cali_get message
[23:37:34]DEBUG - [bambu.rs:1291] MQTT Receive: Print { print: PrintData { ams: None, vt_tray: None, command: Some("extrusion_cali_get"), sequence_id: Some("1"), nozzle_temp_max: None, nozzle_temp_min: None, tray_color: None, tray_id: None, ams_id: None, cali_idx: None, tray_info_idx: None, tray_type: None, reason: Some("fila_device_extrusion_get_all error"), result: Some("fail"), nozzle_diameter: Some("0.6"), filament_id: Some(""), filaments: Some([]) } }
[23:37:34]DEBUG - [bambu.rs:645] -> Message : sequence_id = "1"
[23:37:34]DEBUG - [bambu.rs:694]              extrusion_cali_get message
[23:37:34]DEBUG - [bambu.rs:1291] MQTT Receive: Print { print: PrintData { ams: None, vt_tray: None, command: Some("extrusion_cali_get"), sequence_id: Some("1"), nozzle_temp_max: None, nozzle_temp_min: None, tray_color: None, tray_id: None, ams_id: None, cali_idx: None, tray_info_idx: None, tray_type: None, reason: Some("fila_device_extrusion_get_all error"), result: Some("fail"), nozzle_diameter: Some("0.2"), filament_id: Some(""), filaments: Some([]) } }
[23:37:34]DEBUG - [bambu.rs:645] -> Message : sequence_id = "1"
[23:37:34]DEBUG - [bambu.rs:694]              extrusion_cali_get message
[23:37:34]DEBUG - [bambu.rs:1291] MQTT Receive: Print { print: PrintData { ams: None, vt_tray: None, command: Some("extrusion_cali_get"), sequence_id: Some("1"), nozzle_temp_max: None, nozzle_temp_min: None, tray_color: None, tray_id: None, ams_id: None, cali_idx: None, tray_info_idx: None, tray_type: None, reason: Some("fila_device_extrusion_get_all error"), result: Some("fail"), nozzle_diameter: Some("0.4"), filament_id: Some(""), filaments: Some([]) } }
[23:37:34]DEBUG - [bambu.rs:645] -> Message : sequence_id = "1"
[23:37:34]DEBUG - [bambu.rs:694]              extrusion_cali_get message
[23:37:34]DEBUG - [bambu.rs:1291] MQTT Receive: Print { print: PrintData { ams: Some(PrintAms { ams: Some([PrintAmsData { id: "1", humidity: "2", tray: [PrintTray { id: 0, k: None, cali_idx: Some(-1), tray_info_idx: Some("GFG99"), tray_type: Some("PETG"), tray_color: Some("7C4B00FF"), nozzle_temp_max: Some(270), nozzle_temp_min: Some(220) }, PrintTray { id: 1, k: None, cali_idx: Some(-1), tray_info_idx: Some("GFL95"), tray_type: Some("PLA"), tray_color: Some("2850E0FF"), nozzle_temp_max: Some(240), nozzle_temp_min: Some(190) }, PrintTray { id: 2, k: None, cali_idx: Some(-1), tray_info_idx: Some("GFG99"), tray_type: Some("PETG"), tray_color: Some("161616FF"), nozzle_temp_max: Some(270), nozzle_temp_min: Some(220) }, PrintTray { id: 3, k: None, cali_idx: Some(-1), tray_info_idx: Some("GFG99"), tray_type: Some("PETG"), tray_color: Some("FFF144FF"), nozzle_temp_max: Some(270), nozzle_temp_min: Some(220) }] }]), ams_exist_bits: Some("2"), tray_exist_bits: Some("f0"), tray_is_bbl_bits: Some("f0"), tray_read_done_bits: Some("f0"), tray_reading_bits: Some("0") }), vt_tray: Some(PrintTray { id: 254, k: None, cali_idx: Some(-1), tray_info_idx: Some("GFB99"), tray_type: Some("ABS"), tray_color: Some("FFFFFFFF"), nozzle_temp_max: Some(280), nozzle_temp_min: Some(240) }), command: Some("push_status"), sequence_id: Some("2021"), nozzle_temp_max: None, nozzle_temp_min: None, tray_color: None, tray_id: None, ams_id: None, cali_idx: None, tray_info_idx: None, tray_type: None, reason: None, result: None, nozzle_diameter: Some("0.4"), filament_id: None, filaments: None } }
[23:37:34]DEBUG - [bambu.rs:645] -> Message : sequence_id = "2021"
[23:37:34]DEBUG - [bambu.rs:655]              push_status message
>

alien2108, 2025-03-07:
If printer is switched off, device does not freeze, ping works. The moment it connects to printer ping also stops working.

alien2108, 2025-03-07:
Did you forget to mention microSD card is required? I get following error in console:

[00:02:24]ERROR - [main.rs:383] Failed to load config file '/spoolease.cfg' : Can't open SSD Volume0 - DeviceError(CardNotFound)

yanshay, 2025-03-07:
> Did you forget to mention microSD card is required? I get following error in console:
> 
> [00:02:24]ERROR - [main.rs:383] Failed to load config file '/spoolease.cfg' : Can't open SSD Volume0 - DeviceError(CardNotFound)

Hi @alien2108 

No need for an SDCard. It just allows to provide configuration from file on SDCard in case that's needed for some reason.
 
The web-config error looks like easy to fix, strange it happens only to you for now.

Anyway, can you jump now on discord to try and fix it? Will be easier there.

https://discord.com/channels/1344027434571272252/1344027676461105234

alien2108, 2025-03-07:
I tried the discord link does not work, which server are you on?

yanshay, 2025-03-07:
Try this: https://discord.gg/QY55uSp7

yanshay, 2025-03-08:
I understood from the logs the reason for the first freeze and was able to reproduce it, will fix later today and let you know.

yanshay, 2025-03-08:
@alien2108 I uploaded Patch-3, please check it out.
It should resolve the first freeze issue and you should be able to use SpoolEase. If it works, I'll update the fix in the next release, so let me know.

As for the inability to connect from your device to the OTA, let me know if it still happens. It looks like some specific networking issue that hasn't been witnessed in other installations so will require more troubleshooting to resolve.

alien2108, 2025-03-10:
Installed Patch-3 and yes, it works now! 

OTA check still does not go on....only ota logs are:

[15:32:50]INFO - [ota.rs:80] Resolved DNS for raw.githubusercontent.com [Ipv4(185.199.111.133)]
[15:32:50]INFO - [ota.rs:53] Fetching firmware metadata
[15:32:50]DEBUG - [ota.rs:112] >>>>>> Initiating request to https://raw.githubusercontent.com/yanshay/spoolease-bin/refs/heads/main/bins/ota/ota.toml
[15:32:50]DEBUG - [ota.rs:113] >>>>>>             using IP: [Ipv4(185......)]

yanshay, 2025-03-10:
There's Patch-4 on that page as well, with more logs related to the OTA, can you try it and send me the logs? Can also on Discord

alien2108, 2025-03-12:
Issues soved

yanshay, 2025-03-12:
> Issues soved

Did the OTA searching for firmware work by any chance? For some reason my additional logs did not appear in the logs you've sent, maybe it got hang earlier than the first logs.

What I did notice is that the github certs expired last week. In my case the logs show errors about that. Maybe in your case it just hangs, not sure why it would behave differently but maybe.

I uploaded an upgrade version to https://www.spoolease.io/flash.html (right button, just don't select erase when offerd to).
You can try it and see if that issue is also resolved. Will be happy to know.

alien2108, 2025-03-14:
OTA still does not work. Log:

[22:38:53]INFO - [bambu.rs:1495] Waiting for IP in Bambu Mqtt Task
[22:38:53]INFO - [main.rs:472] Booting from partition 0
[22:38:55]INFO - [spool_tag.rs:156] Initialized Tag Reader successfully
[22:38:55]TRACE - [spool_tag.rs:174] PN532 Firmware Version response: [50, 1, 6, 7]
[22:38:55]INFO - [spool_tag.rs:175] Established communication with Tag Reader (2)
[22:38:55]INFO - [spool_tag.rs:183] Entering wait for tag loop in nfc task
[22:38:55]DEBUG - [spool_tag.rs:190] Waiting for Tag
[22:38:55]INFO - [wifi.rs:337] Connected to WiFi
[22:38:55]INFO - [wifi.rs:340] Waiting for link to be up
[22:38:55]INFO - [wifi.rs:346] Waiting for an IP
[22:38:55]INFO - [wifi.rs:385] .
[22:38:55]DEBUG - [framework.rs:440] Notified on Initialization Completed 1
[22:38:55]INFO - [view_model.rs:472] ------------------------------------------------------------------
[22:38:55]INFO - [view_model.rs:473] Initialization completed successfully
[22:38:55]INFO - [view_model.rs:474] ------------------------------------------------------------------
[22:38:55]INFO - [bambu.rs:1504] From Bambu MQTT - got IP
[22:38:55]INFO - [wifi.rs:390] Still waiting for an IP address
[22:38:55]INFO - [wifi.rs:352] Received IP: 192.168.3.128/24
[22:38:55]INFO - [framework.rs:309] Checking Firmware Version Over the Air
[22:38:55]INFO - [ota.rs:53] Resolving Dns
[22:38:55]INFO - [bambu.rs:1511] No Printer IP configured, discovering Printer
[22:38:55]DEBUG - [bambu.rs:1540] Waiting for SSDP UDP
[22:38:58]INFO - [ota.rs:80] Resolved DNS for raw.githubusercontent.com [Ipv4(185.199.109.133)]
[22:38:58]INFO - [ota.rs:53] Fetching firmware metadata

alien2108, 2025-03-14:
If I try to do "Over the air network upgrade" from the menu on the device I only get this message each time I press the button:

[22:40:50]INFO - [framework.rs:303] Starting Firmware Upgrade Over the Air
[22:40:50]INFO - [framework.rs:303] Starting Firmware Upgrade Over the Air
[22:40:50]INFO - [framework.rs:303] Starting Firmware Upgrade Over the Air
[22:40:51]INFO - [framework.rs:303] Starting Firmware Upgrade Over the Air
[22:40:52]INFO - [framework.rs:303] Starting Firmware Upgrade Over the Air
[22:40:53]INFO - [framework.rs:303] Starting Firmware Upgrade Over the Air
[22:40:53]INFO - [framework.rs:303] Starting Firmware Upgrade Over the Air
[22:40:53]INFO - [framework.rs:303] Starting Firmware Upgrade Over the Air
[22:40:53]INFO - [framework.rs:303] Starting Firmware Upgrade Over the Air
[22:40:54]INFO - [framework.rs:303] Starting Firmware Upgrade Over the Air
[22:40:54]INFO - [framework.rs:303] Starting Firmware Upgrade Over the Air
[22:40:55]INFO - [framework.rs:303] Starting Firmware Upgrade Over the Air

alien2108, 2025-03-20:
All fixed in 0.12! We did it, thnx @yanshay

closed 2025-03-20 by alien2108