yanshay/SpoolEase #17: OTA Update shows error - Failed to read res(op)nse reader

MagicSven81, 2025-03-16:
SpoolEase 0.1.7

if i try to start an update over the air, the above message (including a spelling error) occurs.

A few days ago it works with the message "you have the latest firmware installed"
but now, this error message appears.

yanshay, 2025-03-16:
Yes, I updated on this in Discord, but forgot to update on Github.
The certificates in 0.1.7,  required to connect to the OTA server expired on the SpoolEase device, so unfortunately OTA won't work.
You will need to update now using the upgrade button that was added http://www.spoolease.io (on the right, see image below), using a usb cable connected to a laptop just like you did the first installation.
The certificates are now good for a year and I'll update them when required in future versions.

Make sure to NOT select "Erase Device" (so you won't need to reconfigure everything).

[image]

rtcmay, 2025-03-17:
Discord link? thank you

rtcmay, 2025-03-18:
> > Discord link? thank you
> 
> https://discord.gg/6brKUCERcQ

thank you

closed 2025-03-25 by MagicSven81