yanshay/SpoolEase #19: Initialization failure after flashing and setting up Wifi

ali86x, 2025-03-30:
I tried to setup today.

Got at first the same problem as described here:
https://github.com/yanshay/SpoolEase/issues/18

Erasing by using the webinstaller did not work.
Managed after several times to flash the firmware manually with esptool.js .
Device is booting, opening wifi hotspot and "start page".

[image]
[image]

after connecting to hotspot and entering wifi credentials the device reboots, tries to start and then shows following error:

[image]

also checked the logs on startup, but the only error according missing sd card might be "normal".
also tried to open the webpage (ip shown on startup), but not reachable.

[image]

after that I tried using the option "Upgrade" on the webinstaller and it seems working good (shows "insatllation complete" afterwards).

[image]

but still no connection.

any help or solution?  :(

ali86x, 2025-03-30:
Oh damn, please kill me.
I'm so f**ng stupid :)

I have only seen these error message and did not continue reading the steps for adding the printer.
For sure, the step "swiping down two times" was missing for me.

After that, everything works fine now.
Sorry for that.

Maybe you can add a hint in the manual.
Something like "Please ignore error message after setting up wifi and continue entering printer data" or something.

yanshay, 2025-03-30:
Great to hear!
Will update the docs.

closed 2025-03-30 by ali86x