yanshay/SpoolEase #24: Scale documentation issues

pleasantone, 2025-04-24:
There are some slight improvements in the setup documentation you might consider:

- Probably a Slint vs Mac vs Chrome problem, but I tried "fresh install" on the scale multiple times (like 8 or 9) in order to get to the wifi setup dialog box, and when I finally did (most of the time when I hit next, it returned to the top level "install" dialog box on Chrome), it would not accept legitimate wifi credentials. Finally I solved this by using the Upgrade workflow (since 0.3.0 was installed), and that worked fine. I believe it's getting caught up in the reset logic bringing up the JTAG port vs the main port on the USB-C HID profile.
- You could clarify to use the "upper" USB C port for console access.
- There is one piece I never saw instructions in the build for... on plate 1, there is a cover for the load module PCB, but there is also a smaller piece that also looks like a cover? What's that for?
- The reset sequence between the console and scale seems fragile. I had to reboot the scale to get the console to see it, despite both devices being ping-able and http-able on the same network.

Finally, just a question, once you get out of Alpha, do you intend to do something like a light-pipe for the LEDs? It's fine for now, but as a product you'll probably want something to cover the big hole in the front and expose only the useful USB-C on the side?

In any scale, scale and NFC seem to be working, but I'm a bit under the weather so I didn't test a lot yet.

Again, want to send my thanks, all my comments are ease-of-bringup experiences, the product itself is fantastic and once you get a few more installs by newbies under your belt, I think it will be rock solid.

**Desktop**
 - OS: MacOS current
 - Browser Chrome current

yanshay, 2025-04-24:
Great to hear you got it installed, would be happy to clean out those issues with your help.

> There are some slight improvements in the setup documentation you might consider:
>
> * Probably a Slint vs Mac vs Chrome problem, but I tried "fresh install" on the scale multiple times (like 8 or 9) [...]

I just tried a fresh install and it went smooth for me. Sometimes, due to timing issues on the flashing software, I saw that on first run it doesn't move on directly to the wifi configuration, but then if you press the install button again (not to reinstall, only to configure). 

You will get one of these screens:

[image]

[image]

If the first one pops, you can configure WiFi immediately.
If the second, you can press Logs&Console, go into the logs, and then when you press Back there, the first screen pops up.
It's far from being intuitive, I'll add that better to the documentation.

It's a 3rd party, but optimized for the C++ frameworks and I'm on rust, maybe I'll try fixing it. Or maybe I can get it to work better with the lower USB port.

> it would not accept legitimate wifi credentials

If it allowed you to fill in wifi credentials but did not accept, it means the device tried to connect and wasn't able to. That could be caused by mistyping (which I guess is not the case, but worth making password visible to verify, happened to me a couple of times), or which my guess is here, there was really an issue connecting to your router. It could be WiFi issues - which would align with the issues you saw later, or it could be some incompatibility between the device WiFi and your WiFi router, which I wouldn't rule out. But everything in Scale is the same as in Console from software point of view, so if one connects the other should unless there's some WiFi interference issues.

> * You could clarify to use the "upper" USB C port for console access.

Good Point, I forgot to mention that.

> * There is one piece I never saw instructions in the build for... on plate 1, there is a cover for the load module PCB, but there is also a smaller piece that also looks like a cover? What's that for?

It's mentioned but easy to miss that, I'd make it more noticable:
> Optionally install the appropriate HX711 lid (model includes two options) depending on your connector type

If you solder those pins that come with the HX711 they are vertical, and then when you connect the dupont wires to them they come out vertical. So this lid leaves the ends open for the wires to come out vertically easily. So need to pick the one to use based on what's convenient to use.

> * The reset sequence between the console and scale seems fragile. I had to reboot the scale to get the console to see it, despite both devices being ping-able and http-able on the same network.

That's concerning. Let me know if you see communication issues. I did notice something unclear I think once, and that's when the two devices were nearby, so I thought that was due to WiFi interference.
Let me know if you see that happen again and if you notice this also when the devices are far apart. That needs to be resolved. I made a lot of efforts to make the communication reliable and recover from disconnects (which btw contributed to fixing two rust libraries along the way 😄 )

> 
> Finally, just a question, once you get out of Alpha, do you intend to do something like a light-pipe for the LEDs? It's fine for now, but as a product you'll probably want something to cover the big hole in the front and expose only the useful USB-C on the side?

I'm trying to keep this project to minimum required purchased components and printing something equivalent to a light-pipe from transparent material seemed a bit complex. I don't think white which I could assume everyone have would do well and transparent is not that common.
So the RGB led is for "communicating" with the user and the lower leds are to see liveliness (the power led there is the more important).
If you have any ideas for improving this I'd be happy to hear, just want to minimize the need to find matching components as each one have their preferred sources to source components and it isn't easy to find stuff that's available everywhere.

> 
> In any scale, scale and NFC seem to be working, but I'm a bit under the weather so I didn't test a lot yet.
> 

Hope you feel better soon.

> Again, want to send my thanks, all my comments are ease-of-bringup experiences, the product itself is fantastic and once you get a few more installs by newbies under your belt, I think it will be rock solid.
> 
> **Desktop**
> 
> * OS: MacOS current
> * Browser Chrome current

Let's keep this issue open until I update the docs with everything I put here for others to benefit from.

pleasantone, 2025-04-29:
Sorry I haven’t responded, Covid fog. Will do so asap.

Today I tried to figure out the right order of operations for using both the console and scale with a new spool (and either a new or recycled rfid tag).

The best I could come up with is:
Empty an ams tray location.
Set the filament setting on external slot.
Encode the tag on the console, specifying the external slot.
Load the spool into AMS
Remove the spool and put it on the scale.

This seems wrong, and I suspect not what you intended?

I’m thinking it should be:
1. Empty an ams slot
2. Set filament settings on empty slot (or use ext spool slot)
3. Encode new or used tag using either scale or console scanner
4. select slot you just put parameters on
5. weigh spool on scale
6. Set empty spool weight
7. load spool into an empty slot

but somehow I can’t get this right?

yanshay, 2025-04-29:
Heh, I've got Covid myself now, but not too bad 😄 

Scale nfc scanner doesn't do encoding at this time. I can add it if it helps, but it doesn't right now.
Steps 1-6 seems right, but I don't see where you actually get to perform the encoding. All those steps didn't encode anything yet, this should have been step 7.

Anyway, here is a detailed explanation:

The scale influence the standard encoding process if there's something on the scale while doing a standard encoding.
You could use the external slot, or one of the AMS slots as you prefer.
Let's assume external:
1. Set from Slicer the filament type/color/k/etc. to the external slot.
2. Place a spool on the scale, it should show the weight on SpoolEase console (disregarding encoding or not). (there's also an option to do this step later).
3. Start the encoding process, all slots will flash.
4. Pick the external spool
5. Because there is something on the scale, the weight screen should appear now.  (btw - if you forgot to put the spool on the scale in step 2, and recognize it only now, you can put the spool and it will appear)
6. Now use one of the techniques on screen to pick the spool core weight, go through the process and eventually there will be an Ok button.
7. You should return to the main screen (showing all AMS filaments, with the one you picked flashing, so external in this example) and at the bottom it will say to place the spool tag next to the reader/writer.
8. Now is the time to take the spool off the scale and place it next to SpoolEase Console (note, SpoolEase Scale doesn't encode at this time, maybe it's worth adding that, let me know what you think). It will now be encoded with the spool core weight + potentially the new weight if you asked for that as well. (the value of that would be in future features).

Now, to tell how much is left on the spool, which is currently the value from the scale (more will be added in the future):
1. Scan the spool - information will go into staging.
2. Place the spool on the scale
3. It wil show you, let's say you used some of the filament, 250g /400g - this means out of the total weight of the spool of 400g, 250g is the filament (that's if the core weight entered was 150g).

Let me know if it works as I described.

BTW - in next version this will change a bit, there will be a final screen that summarize what is going to be encoded and allow to manually add some information for capabilities that will be added later.

pleasantone, 2025-04-30:
Replying to old message first...

• Yes, I think the install may have just been a weird timing thing. Both devices have access to the router no problem, and I've been running ESP32 devices around the house for some time. Such is life. Passwords were triple checked, so I can't explain it.
• Thanks for the info on the second cover. I used the longer one because I put a header on the load cell module first. I didn't realize the smaller one served the same function.
• I suspect part of the problem was the two devices were nearby, I moved one to the other side of the desk and things seem to be normal now.
• Thanks for the clarity on spool/scale bringup, I'm going to try it again soon and will report back if I find any weirdness. It's super cool to have the scale nearby.
• If there's any "silly user testing " you want me to perform while you're developing the next version, feel free to poke.

Sorry to hear you've got covid now... sigh, making a resurgence I guess.
Wishing you the best of luck and health!

yanshay, 2025-05-14:
Updated documentation with key points from here. Thanks.

closed 2025-05-14 by yanshay