yanshay/SpoolEase #32: New firmware installation version 0.3.1 fails on new device

BigDi, 2025-05-09:
I started my project today and wanted to flash the new firmware to a new device.
After the flash, I panic. I didn't take a screenshot.
I suspect that something was wrong with the PSRAM configuration.
Then I tried flashing older version 0.2.3. That worked!
Then I tried to update to version 0.3.1, which also worked...

So I don't have a problem and just wanted to note this problem.

BigDi, 2025-05-09:
Could it be possible that the upgrade was not made succesful? 
I can't see any version number in the bottom of the web config and when I enter printer serial and access code then I get an error.

[image]

yanshay, 2025-05-09:
Thanks for reporting the issue.

As for the error you witness now - did you enter security key before trying to apply printer information?

As for the install that failed, could it be that you accidentally tried to install SpoolEase Scale 3.1 on SpoolEase Console? That would fail and will be due to PSRAM issues (different PSRAM configurations of the different devices)

BigDi, 2025-05-09:
May be possible that I have selected the wrong device in the first try.

Now I'm trying to flash again to 0.3.1. After flash I was asked for Wifi but then the screen with secrect will not apear.
I can open configuration but where I can get the secrect?

BTW: Will the nfc tags from the pn532 package work? I get only error and after some retries a unique ID but the display say error in timeout.

yanshay, 2025-05-09:
I went through the process of installing a new SpoolEase Console with 0.3.1 and I can see what could be confusing compared to previous versions, that might be what you stumbled upon that cause the failure to apply settings.

In previous versions you had to activate the web config and there it forced the user to see the security key. Now it doesn't. Still the security key is required for security purpose.
So you need to swipe to the setting screen (two screens below the terminal view), and tap 'Web Config Instructions` to see the security key and fill it in before you fill in printer information.
I'll think how to make this clearer in next versions.

Please let me know if that solved your issue.

yanshay, 2025-05-09:
> May be possible that I have selected the wrong device in the first try.
> 
> Now I'm trying to flash again to 0.3.1. After flash I was asked for Wifi but then the screen with secrect will not apear. I can open configuration but where I can get the secrect?

See my previous answer, we wrote at the exact same time 😄 

> BTW: Will the nfc tags from the pn532 package work? I get only error and after some retries a unique ID but the display say error in timeout.

I received with the pn532 a single card (credit card size). That one won't work. I don't know what you received but likely the same.
The ones that should work are NTAG215 or NTAG216 (NTAG213 may also work but don't have enough capacity).

BigDi, 2025-05-09:
Okay. RTFM next time. I didn't know that I can swipe the screen.
Last time I have entered the secrect and press reset. Now I enterd the secrect and enter the printer setting values without reset. It works!

yanshay, 2025-05-09:
> Okay. RTFM next time. I didn't know that I can swipe the screen.
> Last time I have entered the secrect and press reset. Now I enterd the secrect and enter the printer setting values without reset. It works!
> 

I actually expect people not to read the manual. But didn't notice that behavior change (in prev version you couldn't miss it). 

Happy to hear your recommendations where to place some text on web config so it will be clear that need to do that. That's the easiest fix for me.
I will also see if I can return the web config screen when critical information is missing like at least one printer configured.