yanshay/SpoolEase #39: wrong NTAG size value written to capability container of tag in ensure_tag_formatted

jig432, 2025-05-19:
**Describe the bug**
ensure_tag_formatted (in pn532_ext.rs) writes the wrong value for the number of pages of a tag.  It uses a simple "num_pages" value, when the NXP NTAG spec shows that an encoded value is specified.
In NTAG213_215_216.pdf p. 16 Sec. 8.5.4, Table 4, shows the encodings should be (12h, 3eh, 6dh) for (ntag213, ntag215, ntag216) respectively.
The NTAG203 NXP spec also shows a value of 12h for ntag203.

**To Reproduce**
examine the code line 243:
  let page3_format = [0xe1, 0x10, num_of_pages_on_tag, 0x00]

**Expected behavior**
Correct encoded values should be written to format a tag.

yanshay, 2025-05-19:
Thanks, indeed the size is in 8 bytes blocks and not 4 bytes blocks, fixed in commit da5bd52

closed 2025-05-19 by yanshay

yanshay, 2025-05-19:
BTW - did you encounter a tag that isn't formatted to see this work?

jig432, 2025-05-19:
The only tags I have (ntag215) are pre-formatted, so I can't check it.

For the tags that your code programmed with the incorrect values, I had thought those tags could be reformatted.  However, the spec says (in secc 8.5.4) those bits in page 3 are "set to 1 once", i.e. once a bit is set to 1, it can't be changed. In the code I am developing for mifare tag support, if I find a ntag2xx with a non zero value that isn't one of the ones specified, I'll consider it as the number of pages (to support "backward" compatibility).

The "GET_VERSION" command returns storage_size, but in a strange way and only within a power of 2 size.  I don't have unformatted tags, but if that works even on  unformatted tags, it might be a better way to determine and return tag capacity than trying to probe what data pages are present (and failing for accessing a page beyond what the tag has).  Just a thought.  I thought you had tags that were not formatted that you could try that with.  That command doesn't appear to be defined for 203 tags.

Strangely, I noticed in the nxp ntag spec that the "NDEF memory size" sizes listed in table 4 for ntag215/ntag216 are 8 and 16 bytes shorter than they say the capacity of the tags is (as shown in figures 6 and 7 respectively).  However, statements after table 28 state the sizes are(correctly) 504/888 bytes respectively.

Also, ntag203 tags only have 42 total pages vs. 45 for ntag213.  The last 3 configuration pages of a 213 tag don't exist for a 203 tag.  However, they both have 36 user data pages. 

I  also noticed that 213/215/216 tags (supposedly) are pre-programmed with non zero values in page 4 (as shown in tables 5/6/7).  page 5 of a 213 tag has non zero values also.  At least for my 215 tags (which have the specified non zero values), page 4 can be overwritten with (apparently) arbitrary values.  So I think  starting with page 4 for user data is still safe.