yanshay/SpoolEase #41: Unable to add more than 12 Custom Filaments

iargue, 2025-06-01:
**Describe the bug**
When utilizing the website to upload the mapping for my Custom Filaments name, if I try to add more than 12 custom filaments, I get an error message when applying that just says Error:

**To Reproduce**
Steps to reproduce the behavior:
1. Go to spoolease webpage /config
2. Under Filaments, add more than 12 custom filaments
3. Click Apply

**Additional context**
I did insert an Exfat formatted 128GB sdcard into the device, but no clue if the device is reading that memory card or not. Do we need the sd card to include more than 12 filaments?

yanshay, 2025-06-01:
1. What version are you using? SD card is in use starting the "next version" 3.3 RC4 that needs to be flashed as upgrade at this time.
2. SD card needs to be formatted as FAT/FA32. It can be formatted in the printer which as format function.
3. Can you upload both an image of the filaments and the actual Text so I can try it out? It might be some characters that cause it. There's a limit to the number of custom filaments but more than 12 as far as I remember. The custom filaments should be generated automatically not manually entered, did you do it that way?

iargue, 2025-06-01:
I am running the Spoolease Consolve v0.3.2.

Here is the list of my Custom Filaments.

P3a1d6ae,IID Max PLA Apple Green
Pce8f68d,IID Max PLA Baby Blue
P7441719,IID Max PLA Blue
Pbe52198,IID Max PLA Burnt Orange
P1deb9b9,IID Max PLA Coyote Brown
Pe85a483,IID Max PLA Flourescent Orange
P38ab531,IID Max PLA Fluorescent Green
Pf1a3cf1,IID Max PLA Forest Green
Pdd10c16,IID Max PLA Fuchsia
Paac830c,IID Max PLA Light Pink
Pf0b388c,IID Max PLA Mystic Purple
Pae7b04d,IID Max PLA Red
P62642fe,IID Max PLA Violet
Pd67f91f,IIID Max PLA Pink
P05a4964,SUNLU PETG Basic
P1d5afba,SUNLU PLA black
P4c659d2,SUNLU PLA White

Yes, I generated it automatically through the desktop sync..

Error with 13 filaments.

[image]

No error with 12 Filaments

[image]

yanshay, 2025-06-01:
Confirmed the error happens to me as well, will check and let you know.
Thanks for submitting the issue.

yanshay, 2025-06-01:
I fixed the issue. 
To install a patch go to https://debug.spoolease.io/debug.html
The bottom most button at this time stating "RC4-Patch1" will flash an upgrade. 
Make sure NOT to check the erase, so all configuration will stay.
Then try it and let me know if it work.

yanshay, 2025-06-03:
The firmware you flashed `hd2-1` includes the fix. To me it worked with tens of filaments, about 1500/2000 chars showing below the field. 
Did you try with more than that? If so how much? If not, then please post the filaments you tried here, maybe now it's another issue now.
Did you still see Error saying nothing?

iargue, 2025-06-04:
I tested it today and had no issues saving the filament.

yanshay, 2025-06-04:
Great, so this one is solved.