yanshay/SpoolEase #43: Add integration with spoolman

maelremrem, 2025-06-08:
**Feature request**
Can you add the support or integration with Spoolman ?
[SpoolMan GitHub](https://github.com/Donkie/Spoolman)

This spool management software is great and will be a great addition to this project.

yanshay, 2025-06-08:
Thanks for the input.

I have a few major features in the pipeline after which I will probably look into integrations.

I am not that familiar with Spoolman other than it being a database holding a list of spools. Does it do more than that?

Can you tell me what use cases it has other than storing list of spools and allow searching/filtering/location etc. 

It seems to be a pretty large project so it must be more than just what I know it does.

somerockenguy, 2025-06-15:
I use spoolman to manage my inventory of filament with QR tags stuck to the side of the spools. It is self hosted on a virtual machine and gives me a list of all filaments I have and how much is remaining on the spool. With bambu printers it requires manually weighing the spools and updating the remaining amount, but I believe there are plugins for octoprint to allow for this to be handled automatically when using marlin or klipper based printers.

Ideally I would like to the spools scanned via NFC with SpoolEase to be able to be weighted with SpoolEase Scale and update the database I have already built in Spoolman.

I am only using bambu printers currently so SpoolEase implementing its own inventory system would help workflow but some users with a mix of bambu and marlin/klipper printers would be looking for a way to track filament used across all printers.

yanshay, 2025-06-15:
Currently encoding tags with SpoolEase does require Bambulab printers but I plan to allow that to be independent of the printer and then anyone will be able to encode tags and use the Scale and the inventory system and more that will come later on as part of the system. Even thought of having a standalone device for only weight + inventory + additional features, so w/o the Bambulab specific features, that would be very relevant to the entire 3D printing community that don't use Bambulab printers.

Adding integrations might arrive later, but integrations, and bugs in integrations when I don't control the other party and version changes, backwards compatibility, will be a lot of work to maintain and more importantly to support, so I prefer to push it to after I feel the other major features are introduced and I get to mostly incremental small features, since each might require me the break the integration and change how it is working.

A question I have - why is it that only with Bambu printers it (spoolman) requires manually weighting the spools and updating the remaining amount? Is it because marlin/klipper tracks how much filament is consumed while printing? Does it support properly also print failures?

I don't fully trust these automated approaches because if filament breaks it isn't aware of that, or if spool doesn't have exactly the advertised weight you have to use a scale anyway, and the extra effort of scanning and weighting after printing is a small price to pay for a reliable weight.

tugsi, 2025-07-25:
There's a German-language project that weighs and transfers the data to Spoolman, where it also retrieves it.
https://github.com/ManuelW77/Filaman

Spoolman also has a huge database with data on existing commercially available filaments and their specifications.

Spoolman works with an API, so the risk of functionality being lost after updates is lower.

I would welcome an integration, but I also understand your concerns.

A quick note on the topic of weighing from the system: as far as I know, it's only an estimate of how much filament the splicer indicates. So, it's not accurate, but it's better than nothing.

yanshay, 2025-07-25:
Thanks for your feedback.

Since I answered on this thread SpoolEase has progressed quite a lot and includes a built in inventory management which is in alpha right now.
While I could try to build on an external project like SpoolMan for that I wouldn't be able to deliver some of what I already implemented there at the level I want to and also I wouldn't be able to deliver what I will soon release w/o having to develop code inside SpoolMan itself and also it would be less reliable under some circumstances due to network gap and potential network issues. It would also be much more difficult to setup to the non-tech user (having to deal with docker, etc.).

Also, if I add integrations, maintaining them would slow me down considerably introducing new features since integration is something one must maintain and keep backward compatibility etc.

So while in the future I may add such integration it won't be able to replace the internal functionality I have to develop but will slow me down on the core features I want to implement.

I prefer to have the project deliver a solution that provides the best experience to the user as a whole rather than having every component best in class but the integrated solution not so and hard to maintain.

As for slicer weight being inaccurate - it really depends on the density setting for the specific filament. The slicer's length metrics I believe are quire accurate, otherwise 3d printing wouldn't work and we wouldn't get such accurate models. However converting that to weight is a simple math calculation that depends on the filament density setting. If that's not filled in correctly it would be inaccurate. Since many just use Generic or Bambulab filament settings for everything even if that's another vendor then that ends up being inaccurate but I don't think that much.

As a side note, I looked at SpoolMan data when it came to empty spools weights, and the data there was just incorrect and even on very common spools types who's weights are widely known (I think it was Bambulab's, not sure), so while it may contain a lot of data I'm not sure it is accurate, at least not when it comes to empty spool weights.

yanshay, 2025-07-25:
Also, I hope to release soon what @somerockenguy mentioned that's available for marlin/klipper, only for Bambu printers. That's in addition to manual weighting already available, and so SpoolEase users will enjoy both techniques for both easy and accurate weight measurement.

tugsi, 2025-07-26:
Thank you for your detailed answer, and I completely understand.
As a developer, you have a certain idea, and the project is alive and in flux, so peripheral noise is rather distracting.
Is there some kind of API on your side planned? Or a JSON that outputs data from the currently weighed spool (filament type, weight, etc.) so that everyone would have the opportunity to use it elsewhere?
I previously had a test setup with the Filaman project, but I like your project better. I think there's more potential here. So I'm currently building it, but I realized that the hole spacing of my scale modules is smaller than the ones you use. So I'll order the right ones from Ali, but first I'm going on vacation. Thumbs up for your work.

yanshay, 2025-07-26:
There's currently no API mostly due to constant changes between versions (internally I already has some backwards compatibility between versions).

Once I feel it stabilized I may add ones. 

It would help if in the mean time you can share what integration are you looking to do. Maybe that's already on the roadmap of what I plan to develop. If not maybe will be easier for me to add it rather than support integration and could arrive earlier.

tugsi, 2025-07-30:
I don't have a specific use case at the moment. I'm already running a Spoolman version because of the other scale project and would like to continue using it. I've only been testing it so far, so it doesn't hurt if I let it die.

I just thought that if there's already a very successful inventory project, there's no need to reinvent it.

I haven't looked at your alpha yet, though.

My thought was that if there's a JSON output, for example, you could create almost anything with it to integrate it into various other projects.
If I have anything concrete, I'll let you know.

closed 2025-10-17 by yanshay