yanshay/SpoolEase #48: Inventory ease-of-use features

pleasantone, 2025-06-13:
[] Backup/restore of the inventory database (using file download feature and file upload) -- .json would be a great format.
[] Export function for the inventory database - different as it would be .CSV
[] Delete row button in /inventory screen
[] Clear all rows button in /inventory screen
[] Make password UI intuitive (allow for a button)
[] Format SD card (?) perhaps...

These are all nice-to-haves, feel free to ignore any/all of them.

yanshay, 2025-06-14:
> Backup/restore of the inventory database (using file download feature and file upload) -- .json would be a great format.

The backup format will still change, but the purpose is not to be used but for restore, so the format is built for easy of restore.

> Export function for the inventory database - different as it would be .CSV

What do you mean here?

> Delete row button in /inventory screen

I will add

> Clear all rows button in /inventory screen

I will add

> Make password UI intuitive (allow for a button)

I still plan to modify this, but do you prefer pressing an extra button if it can be automatic?

> Format SD card (?) perhaps...

This is unfortunately not supported by the library I'm using so would be too complex to add.

yanshay, 2025-06-17:
alpha4 now includes delete

https://debug.spoolease.io/inventory.html

closed 2025-08-17 by pleasantone