yanshay/SpoolEase #50: Ability to code tags outside of what is loaded in Bambu Studio, such as via a mobile app

mytechguyri, 2025-06-28:
One of the things I've run into with my A1 Mini is, there are certain filaments that it won't let you select in Bambu Studio, such as ASA, or PPA-CF.... so you can't encode them by adding to Bambu Studio. 

Give the user the ability to encode a tag either through the web interface, or a mobile app with brand, type, sub-type, color, etc.

yanshay, 2025-06-28:
Bambustudio doesn't allow to use ASA on A1 Mini? Interesting.
It probably won't work well as is but what if someone puts the A1 mini in an enclosure?
I think I read somewhere there is an option to remove all kinds of built in constraints that may allow that to work as well.

Side note: if you have A1 why do you want to use ASA? For another printer?

Anyway, this is coming as part of inventory management feature, I'm close to completing this specific feature that will allow encoding without selecting a slot as base of the info. 
It might actually let you also try printing ASA on A1.

mytechguyri, 2025-06-30:
I have oth a P1S and an A1 Mini.... Just sometimes the P1S is in use on a larger print, and I don't want to wait... the Mini will do it (especially with a 80C cryogrip glacier plate), just Bambu doesn't officially support it.  Same with PPA-CF... actually the mini prints that better than the P1S does, but Bambu doesn't officially support it, so only if I force the filament type with an open RFID device (i currently have an openspool device, but am buidling the spoolease now) then it accepts PPA-CF just fine and lets me print with it.

yanshay, 2025-08-21:
Should be possible to do with Inventory in the Alpha version

closed 2025-08-21 by yanshay