yanshay/SpoolEase #52: Integration with Ajax3D TD-1/BIQU AjaxTD1S

mytechguyri, 2025-07-12:
As you are likely aware, the Bambu printers will accept an alpha channel value from MQTT when passing AMS colors to the printer.. i.e. #RRGGBBAA    This is really useful for transparent or translucent filaments.

The TD-1 device automatically gives you the RGB hex color value, and a HueForge Transmission Distance, which with alpha = 255 × (1 - ((TD - TD_min) / (TD_max - TD_min))) can be converted to an alpha channel value.

Would be great, when adding a new filament to have some sort of connectivity to a TD-1 device to automatically read the filament color and TD value to get the RGBA color and alpha channel values.
(the creator of the TD-1 has a build it yourself BOM, so this might be something could even be integrated right into the SpoolEase hardware directly)

mytechguyri, 2025-07-12:
So i was brainstorming this a bit more with the help of AI, and this could be something simply integrated into the SpoolEase device just by having a 2mm passage to pass filament through that would put the filament in front of a TCS3472 RGB sensor and a SK6812 RGBW neopixel, so no nedd for the Ajax TD-1 at all... just a simple filament color sensor 

AI came up with the following code to make it all work.... sorry, I don't code in C++ so I don't know if this is garbage code or not, but figure this might be a good start for this enhancement... this assumes a constant reading, but I'd think a simple microswitch to detect if filament inserted (like an old school filament runout sensor) could call this routine whenever a new spool is being added and filament is detected as inserted into the RGB detection hole.

#include <Wire.h>
#include <Adafruit_TCS34725.h>
#include <Adafruit_NeoPixel.h>

// --- NeoPixel Setup ---
#define LED_PIN       5
#define LED_COUNT     1
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRBW + NEO_KHZ800);

// --- TCS3472 Sensor ---
Adafruit_TCS34725 tcs = Adafruit_TCS34725(TCS34725_INTEGRATIONTIME_700MS, TCS34725_GAIN_16X);

// --- Reference Clear Value ---
const float reference_clear = 14000.0;

// --- Averaging Settings ---
const int samples = 10;
const int sample_delay_ms = 100;

void setup() {
  Serial.begin(115200);
  Wire.begin();

  if (!tcs.begin()) {
    Serial.println("Couldn't find TCS34725");
    while (1);
  }

  strip.begin();
  strip.setBrightness(150);
  strip.show();

  // Turn on Natural White LED for neutral lighting
  strip.setPixelColor(0, strip.Color(0, 0, 0, 255));
  strip.show();

  delay(1000); // Allow lighting to stabilize
}

void loop() {
  // --- Averaging Variables ---
  uint32_t sumR = 0, sumG = 0, sumB = 0, sumC = 0;

  // --- Capture Multiple Samples ---
  for (int i = 0; i < samples; i++) {
    uint16_t r, g, b, c;
    tcs.getRawData(&r, &g, &b, &c);
    sumR += r;
    sumG += g;
    sumB += b;
    sumC += c;
    delay(sample_delay_ms);
  }

  // --- Calculate Averages ---
  uint16_t avgR = sumR / samples;
  uint16_t avgG = sumG / samples;
  uint16_t avgB = sumB / samples;
  uint16_t avgC = sumC / samples;

  // --- Normalize RGB ---
  float total = avgR + avgG + avgB;
  uint8_t red   = (uint8_t)(255.0 * avgR / total);
  uint8_t green = (uint8_t)(255.0 * avgG / total);
  uint8_t blue  = (uint8_t)(255.0 * avgB / total);

  // --- Derive Alpha from Clear ---
  float alpha_f = 255.0 * (1.0 - ((float)avgC / reference_clear));
  uint8_t alpha = constrain((uint8_t)alpha_f, 0, 255);

  // --- Format Hex RGBA ---
  char hexString[11]; // "#RRGGBBAA\0"
  sprintf(hexString, "#%02X%02X%02X%02X", red, green, blue, alpha);
  
  Serial.println(hexString);

  delay(2000); // Pause before next scan loop
}

yanshay, 2025-07-12:
It's definitely an interesting idea and a good match for SpoolEase. I might try to take a look at something like that in the future.

I think the complexities are in accuracy of results, calibration, etc. I read some criticism on TD-1 results, especially the color. 

Seems like something I'll want to at least play with when I get some time.