yanshay/SpoolEase #55: Scale Divide by Zero after console connected

StevenBonePgh, 2025-08-06:
**Describe the bug**
I just printed and set up spoolease with scale (thank  you!). I purchased a 5kg load sensor. After initial configuration of the scale all was well (red light) until I plugged in the console. When the console and scale connected the scale's red light would go out. Checking logs on the scale showed the following: 

**Additional context**
[08:28:46]INFO - [ota.rs:66] Current firmware version 0.3.1 is up to date
[08:29:24]INFO - [web_server.rs:323] 0: Connected from Some(Endpoint { addr: Ipv4(), port: })
[08:29:24]INFO - [load_cell.rs:264] Initializing Load-Cell reader
[08:29:24]INFO - [load_cell.rs:278] Load-Cell reader initialized successfully
[08:29:25]INFO - [console_proxy.rs:172] SpoolEase Console connected
[08:29:26]
[08:29:26]
[08:29:26]====================== PANIC ======================
[08:29:26]panicked at src/load_cell.rs:141:30:
[08:29:26]attempt to divide by zero
[08:29:26]
[08:29:26]Backtrace:
[08:29:26]
[08:29:26]0x4205becb
[08:29:26]0x4203d32a
[08:29:26]0x4203353d
[08:29:26]0x4206452b
[08:29:26]0x420e7dca
[08:29:26]0x40378ef0

yanshay, 2025-08-06:
A few questions:
1. Does it happen every time?
2. Did you calibrate the scale (not sure you had the chance).

Looking at the code, the only way I can see this hapening if something went wrong in the calibration process. 
Most likely a case where the load cell is not functioning properly, and you did calibration.
Did you see as in the setup guide that **before** clibration, when you place something on the scale you see the values changing on the top of the calibration screen? If it's a constant number it would result in this panic.

StevenBonePgh, 2025-08-07:
I came to the same conclusion as you did as to the div zero cause.  I checked to see where the calibration values were stored, and saw it was on the scale. May be a good idea to log those values for troubleshooting. Anyway, I erased and reinstalled on the device and started from scratch and no issues.

As to the symptoms for future folks, it does happen every time the console connects to the scale. The scale has the solid red light that it is connected to wifi. When you power up the console and it connects to the scale you will see the red light go out when the fault happens.

As for the pre-check, yes, the numbers changed. I wonder if somehow I fat-fingered the tare button while the weight was on the scale. I think it essentially calibrated then immediately died, as after calibration, the main screen no longer showed the scale. After an erase and reinstall, it worked as described, where the scale showed the weight immediately after calibration.

closed 2025-08-07 by StevenBonePgh

yanshay, 2025-08-07:
The strange thing here is that both the tare value and calibration weight value were exactly the same to get a division by zero.

These numbers in the code are the load cell values which to get two samples exactly the same would be close to winning the lottery. It's not grams, it's a large noisy number.

So either you were very (un)lucky or something was faulty during the calibration time and same value returned from the hx711. So notice if you see any other strange behavior.

Or maybe you stumbled on some bug there under certain conditions.

That said, the calibration process is very minimal, I didn't want to invest in it and preferred to spend time on more significant features. Maybe with time I will get to improve it.